corna/me_cleaner #353: Unknown image checkin Asus N752VX rom file

Xwang1976, 2021-03-10:
When I run:
me_cleaner -c N752VXAS.202.rom 
I get the following message:
Unknown image

The N752VXAS.202.rom is obtained using this process:
1) download the zip file from ASU website: https://dlcdnets.asus.com/pub/ASUS/nb/N752VX/N752VXAS202.zip
2) unzip it to obtain the N752VXAS.202 file
3a) run dd if=N752VXAS.202 of=N752VXAS.202.rom bs=1024 skip=2 
or alternatively 
3b) use uefitool to extract the body saving it as  N752VXAS.202.rom
(the file obtained is the same according to diff)

Is there anything wrong in what I've done?
Why the image is not recognised?

Stitch626, 2021-07-10:
Yea, there is something wrong.

You can't use the images provided by asus. Those are incomplete. Dump the chip and use the 8MiB dump as base.
BTW: You will most likely be unable to flash/dump the bios without an external flasher. ASUS is somewhat special there... (I had to learn it the hard way with a bricked device)

czeej, 2021-07-17:
@Stitch626 
Please explain what you know about ASUS UEFI? 
I have had good luck so far with Asus motherboards but maybe. It's not as good as I think.

Stitch626, 2021-07-17:
To answer both your original kinda question and the edited one: it depends on your hardware.
Some (most?) ASUS desktop boards get a kinda full image on the asus website, it's just encapsulated. The laptops are different.
ASUS tends to use some kind of "special" flash-tool. It reads some critical areas off the currently active bios (for example the serial number, model typ, windows key, etc), saves it, and uses the "reduced" flash file you just downloaded with the saved information. Some flash files don't even contain the IME part and leave it as it is.
The given error message seems to be related to some ME-internal checks against altered images.
Some vendors use the ME firmware's update tool. Depends often on the given type of hardware.

To explain it on a example:

Let's say you start a bios update. The update tool will happily flash the bios region of the chip, and loads the ME image into the  RAM (or a secondary bios-chip, again, depends) and once you reboot the machine and the bios starts its procedures, the ME firmware update will be triggered by a command. But it's done by the ME processor itself, and thus verified.
The whole image will be checked. That sounds like the error message you got. 
Hope that explained it just fine :)