corna/me_cleaner #355: Help to make a valid backup of flash chip

ginto37, 2021-03-26:
I can't make a valid backup of the flash chip on my Lenovo x230. I'm using SOIC clip and `flashrom`. The chip is in-situ, not de-soldered. Here's what I've tried:

- shortened the cables as much as I can (I think 10cm now)
- re-seated the SOIC clip
- checked my GPIO configuration
- powered the chip through the SOIC clip and the Wake-On-LAN methods
- reduced the `flashrom` read speed.

Multiple dumps of a single configuration will match (I compare the checksums) but they always fail the `ifdtool` check and the `me_cleaner` test:

- `ifdtool` - some regions will always fail the read/write access to themselves.
- `me_cleaner` - it always returns `Unknown Image`

The laptop works perfectly so I don't think the flash chip is faulty. Can someone make a suggestion?

skochinsky, 2021-03-26:
You're probably dumping the wrong chip. AFAIK x230 uses two flash chips, one for IFD+ME, another just for the BIOS.

https://blog.noq2.net/corebooting-thinkpad-x230.html

ginto37, 2021-04-02:
Igor Skochinsky:
> You're probably dumping the wrong chip. AFAIK x230 uses two flash chips, one for IFD+ME, another just for the BIOS.
> 
> https://blog.noq2.net/corebooting-thinkpad-x230.html
> 

I'm definitely dumping the correct chip. It's the upper one furthest 
from the front of the laptop. My SOIC clip should be attached the right 
way too, it's easy to orient it because of the physical dot on the 
bottom right of the chip.

The page you link to has a link to another walkthrough, where it says:

> Make sure the pinouts are correct; otherwise, Flashrom will fail to detect a chip, or it will "detect" a 0x0 chip. 

In my case `flashrom` detects the chip correctly, no doubt about it.

> Finally, make sure that the Pomona clip makes contact with the metal wires of the chip. It can be a challenge, but keep trying.

My SOIC clip seems to be be making contact with all the pins but I've 
only been able to verify this visually. What error would I get if my 
SOIC clip wasn't seated correctly?

skochinsky, 2021-04-02:
Try dumping the other chip anyway. How big is the dump? The IFD should have 5A A5 F0 0F at offset 0x10 and recognized by UEFITool as "Intel Image".

ginto37, 2021-06-11:
I finally tried dumping the other chip but I couldn't get `flashrom` to recognize it:

> `No EEPROM/flash device found.`

There is no information stamped on this chip like the other so I have no clue what chip info to pass to `flashrom`. I tried the same chip info as the first chip but this doesn't work.

I tried UEFITool on a dump from the first chip. I get:

> `Type: Image`
> `Subtype: UEFI`
> `Size: 00400000`

The size reported in `ifdtool` is 4194304 bytes.

Where do I find that IFD offset info? What tool should I use for that?

czeej, 2021-07-17:
My experience with Dell dumping the 32 bios chip. Was tricky. It's like it didn't like to be read. 
You manually enter the chip name in flash rom. 
use the -c and chip name usually you just read it if you can or change your 64 to 32 in the name. 
It may read once you do that.

ginto37, 2021-08-20:
Hi czeej,

Thanks for the tips. I think my problem is just identifying the 
Make/Model of the 8MiB chip. I found a list online of all the chips 
targeted by previous Lenovo firmware updates for the X230, I just didn't 
have time to try them yet. I got tired of taking my X230 apart so many 
times! :)

If I still have problems I'll start looking at the things you've 
suggested. I think my SOIC clip is good though, and the chip pins look 
shiny.

By the way, do you know how to get the offset information that 
skochinsky was talking about above?

czeej:
> My experience with Dell dumping the 32 bios chip. Was tricky. It's like it didn't like to be read.
> You manually enter the chip name in flash rom.
> use the -c and chip name usually you just read it if you can or change your 64 to 32 in the name.
> It may read once you do that.
>

czeej, 2021-08-25:
You may have to ask Skochinsky, he is more or less an expert here.
5A A5 F0 0F is probably a binary / hex. at offset 0x10 as the address location
You use bin walk or other software to analyze the image like UEFI tool. 
the 5A A5 F0 0F could be a header to show you intel's code follows. 
I honestly, do not know. I am guessing. that seems to be a way to ensure you have the right image with IME firmware