corna/me_cleaner #356: You need to be root.

Dmole, 2021-03-28:
The instructions here

https://github.com/corna/me_cleaner/wiki/Get-the-status-of-Intel-ME

don't work;

```
# ./intelmetool -m
iopl: Operation not permitted
You need to be root.

# id
uid=0(root) gid=0(root) groups=0(root)

```

Dmole, 2021-03-29:
> intelmetool is not part of this project

Correct; this project should not link to a broken project with a closed issue tracker.

frantic-from-paranoiawf, 2024-12-03:
> > intelmetool is not part of this project
> 
> Correct; this projects wiki should not link to a broken project.
> 
> https://ticket.coreboot.org/issues/302

This is a user error. Links another project as well, blames me_cleaner. Such stupidity.