corna/me_cleaner #357: The ME/TXE region is valid but the firmware is corrupted or missing

Gert-Vanhaerents, 2021-04-01:
I try to clean or disable intel ME but when i do the needed procedure:
`$ python me_cleaner.py -S -O modified_image.bin original_dump.bin`
I have the following error:
```
Full image detected
The ME/TXE region is valid but the firmware is corrupted or missing
```

I have tried the original bios from the manufacturer and also a bios backup made with the bios flashing tool. I don't have a programmer (i orderid some one but it take some time to deliver).

How can i backup the installed bios? or how can i use the manufacturers bios to do the me-cleaner?

The laptop is a Clevo NL51CU with a Intel i5-10210U cpu 10th generation.

theordinarymanpure, 2021-07-07:
+1
Hi there. I'm having the same issue.

I have an Intel NUC (10th-gen 10510u) box.

I've been following this guide (https://github.com/corna/me_cleaner/wiki/External-flashing) 

I was able to read the ROM contents:
`Found Macronix flash chip "MX25L25635F/MX25L25645G" (32768 kB, SPI) on linux_spi.`
until having been stuck at this point: "Sometimes ifdtool prints a resonable output even in case of an invalid image (like this)".
In my case it was lacking RW permissions in the referenced part of firmware dump.

`Found Master Section
FLMSTR1:   0x00a00f00 (Host CPU/BIOS)
  Platform Data Region Write Access: disabled
  GbE Region Write Access:           disabled
  Intel ME Region Write Access:      disabled
  Host CPU/BIOS Region Write Access: disabled
  Flash Descriptor Write Access:     disabled
  Platform Data Region Read Access:  disabled
  GbE Region Read Access:            disabled
  Intel ME Region Read Access:       disabled
  Host CPU/BIOS Region Read Access:  disabled
  Flash Descriptor Read Access:      disabled
  Requester ID:                      0x0f00

FLMSTR2:   0x00400d00 (Intel ME)
  Platform Data Region Write Access: disabled
  GbE Region Write Access:           disabled
  Intel ME Region Write Access:      disabled
  Host CPU/BIOS Region Write Access: disabled
  Flash Descriptor Write Access:     disabled
  Platform Data Region Read Access:  disabled
  GbE Region Read Access:            disabled
  Intel ME Region Read Access:       disabled
  Host CPU/BIOS Region Read Access:  disabled
  Flash Descriptor Read Access:      disabled
  Requester ID:                      0x0d00

FLMSTR3:   0x00800900 (GbE)
  Platform Data Region Write Access: disabled
  GbE Region Write Access:           disabled
  Intel ME Region Write Access:      disabled
  Host CPU/BIOS Region Write Access: disabled
  Flash Descriptor Write Access:     disabled
  Platform Data Region Read Access:  disabled
  GbE Region Read Access:            disabled
  Intel ME Region Read Access:       disabled
  Host CPU/BIOS Region Read Access:  disabled
  Flash Descriptor Read Access:      disabled
  Requester ID:                      0x0900`

The former of the two ifdtool checks ("the region sizes in the FLREGn section make sense") looks fine.

But the me_cleaner does not recognize the dump-ed ROM image:

`python /home/pi/Documents/git/me_cleaner/me_cleaner.py -c original.rom
> Full image detected
> The ME/TXE region is valid but the firmware is corrupted or missing`

So would anyone please give a hint, should I proceed with cleaning this ROM image with me_cleaner?

I'm attaching the full ROM image for better understanding. I won't use this NUC box anyways without ME disabled.
[original.zip](https://github.com/corna/me_cleaner/files/6777612/original.zip)

Thank you in advance

Stitch626, 2021-07-10:
Hi,
I guess me_cleaner is just a bit outdated. I have checked your file with UEFItool, UBU and ME Analyzer. Everything seems to be fine. ME version 14.

I don't know if current platforms still use the HAP bit, but I'd only go with that (soft mode). Me_cleaner may work regardless of the error message. You will have to test it as corna isn't really active anymore. 
But be prepared to unbrick your device in case everything goes wrong...

theordinarymanpure, 2021-07-11:
> Hi,
> I guess me_cleaner is just a bit outdated. I have checked your file with UEFItool, UBU and ME Analyzer. Everything seems to be fine. ME version 14.
> 
> I don't know if current platforms still use the HAP bit, but I'd only go with that (soft mode). Me_cleaner may work regardless of the error message. You will have to test it as corna isn't really active anymore.
> But be prepared to unbrick your device in case everything goes wrong...
> 
> Edit: #304

Thanks, Stitch626

I already figured that out. 

I tried the [dt-zero](https://github.com/dt-zero)'s fork of me_cleaner (https://github.com/corna/me_cleaner/pull/282) and it didn't work neither. In my case I was able to apply me_cleaner forked tool, but the Intel NUC became bricked. Though when I flashed the original rom image it launched as usual. Seems like in my case there is an Intel Boot Guard or alike is enabled.

BTW, as per ValoWaking's comments here (https://github.com/corna/me_cleaner/issues/349) there seems to be no place to escape to. Even Raspberry Pi boards presumably have hardware backdoors (https://linustechtips.com/topic/1242648-does-raspberry-pi-have-a-backdoor-like-intel-me/).

Does anybody know any platform - even Raspberry-grade -  without hardware backdoors it them?

Stitch626, 2021-07-11:
You may read something here: https://libreboot.org/faq.html#intelme
As both my and ValoWaking's comments lack a lot of information.

But in the end, there is no escape at all. You can only get *some* freedom back. 
It seems like x79 and probably other older chipsets have no hardware watchdog inside, which means you can remove the IME code entirely out of the flash (as I did on my x79 Board) and be happy you removed something.

It gets more complicated on wireless stuff, as most of it is proprietary. And ARM? It' s even worse. 
I did indeed already know that most arm based chips have some kind of firmware. What a surprise, huh? 
Same for coprocessors and stuff. 

If you are paranoid enough and need x86, go the Libreboot path, or choose older chipsets which have no hardware watchdog on removed ME FW.
Otherwise you may take a look at RISC-V (as mentioned in the LTT thread) or look deeply for older ARM chips. 

But I personally would'n mind enough. Disabled IME on older chipsets is enough since even LED lights have electronics and some logic circutry inside...

Our life is digital and contains a lot of closed firmware. You either deal with it or run away and live in the woods :D

theordinarymanpure, 2021-07-11:
> You may read something here: https://libreboot.org/faq.html#intelme
> As both my and ValoWaking's comments lack a lot of information.
> 
> But in the end, there is no escape at all. You can only get _some_ freedom back.

That is completely unacceptable.

It discourages doing anything valuable knowing it could just be stealed by those Levys pedos.

Seems like the hard way is the only one left.

Stitch626, 2021-07-11:
Everything can be stolen. Doesn't matter if its digital or analog. Only your brain belongs to you. Yet.

But... research tries to change even that.  :>