corna/me_cleaner #359: me_cleaner and Synology DS1819+ (Intel Atom C3538) NAS

gmipf, 2021-06-16:
If I have understood correctly, the Intel Atom C3000 platform also should contains Intel ME. Did somebody ever tried to do the removal on one of those Synology embedded platforms? Or is it even physically possible on this NAS model? Here are photos of the mainboard: https://www.techpowerup.com/review/synology-ds1819plus/4.html

The flash type is NAND, I should be able to dump that type of memory.

skochinsky, 2021-06-17:
the BIOS is highly unlikely to be on NAND; it's probably in the Winbond SPI chip.

Stitch626, 2021-07-11:
Ye it is. Those devices have just a kinda regular UEFI like usual PCs. But I couldn find a dump of it. Create a dump and take a look at it with UEFItool. If it shows any kind of ME region, it MAY be possible.

jamesy0ung, 2021-07-30:
[image]
This is the chip you want.

gmipf, 2021-07-30:
Thanks but I already discarded the idea of buying a Synology.