corna/me_cleaner #362: my UEFI already has an option to disable ME. Does this make me_cleaner redundant?

KSPAtlas, 2021-08-04:


ginto37, 2021-08-20:
Depends. me_cleaner can completely remove ME from your device, which is 
safer because it can't be re-enabled by a script or update later without 
you knowing. But completely removing ME might cause problems for your 
device.

You could try disabling in UEFI and then check the status of the ME 
regularly using the information in the wiki, just to make sure it's 
effective and stays disabled: 
https://github.com/corna/me_cleaner/wiki/Get-the-status-of-Intel-ME .

KSPAtlas:
> 
>

davidhealey, 2021-09-06:
> Does this make me_cleaner redundant
How much do you trust your motherboard manufacturer? UEFI is already proprietary so there is no way to confirm what the disable ME option they provide actually does.

FrostKnight, 2022-02-06:
Not even close, uefi is crap in general anyways...

Btw, without coreboot, I don't see the point of me cleaner, if the device is new enough, due to intel's evil ways, they might find a way to make it work again. Aka, the intel me. :(

KSPAtlas, 2022-02-06:
> Not even close, uefi is crap in general anyways...
> 
> Btw, without coreboot, I don't see the point of me cleaner, if the device is new enough, due to intel's evil ways, they might find a way to make it work again. Aka, the intel me. :(

Its an Intel b75 motherboard with an Intel i7 2600. So not exactly new.

FrostKnight, 2022-02-06:
> Intel i7 2600

Hmm... are you talking about this one?

https://www.intel.com/content/www/us/en/products/sku/52213/intel-core-i72600-processor-8m-cache-up-to-3-80-ghz/specifications.html

I thought he must have meant one that was like gen 4 or higher...

Perhaps there is a purpose then... 

Hmm... 

Yeah, I don't see why it wouldn't be a good idea then.

I thought it must have had some irritating bootguard and proprietary graphics and sound blob requirements.

Yeah, it is worth it then.    My bad... 

Had no idea...

Espionage724, 2023-07-13:
I don't believe it does, but I guess it could vary depending on the implementation.

On the Dell Latitude 5591 I have (Coffee Lake), there's a EFI variable for disabling ME (can set with setup_var), but it doesn't do anything. Using me_cleaner to set the HAP bit does disable ME though.