corna/me_cleaner #370: Issue with iGPU on Macbook Pro 9,2 (i7-3520M)

dandongus, 2021-12-23:
After running me_cleaner (with -S) on a dump of the uefi firmware and flashing the output back to the eeprom, the machine boots up fine and the ME is disabled/removed, but there's no "graphics acceleration" in OSX, it's instead running with just software rendering. Flashing back the original image restores full functionality, as one would expect.

Anyone else have a similar issue or any suggestions for what to try?

490398290, 2022-01-21:
macOS won't work without ME firmware. So, if you want to have hardware acceleration i suggest to flash back original firmware.

closed 2023-01-20 by dandongus

Espionage724, 2023-07-13:
ME handles protected audio/video playback for Intel graphics (at least on some platforms, if not all), so if macOS has a hard requirement for HDCP that might be an issue.

> macOS won't work without ME firmware

Do you have any more details about this? What does macOS care about ME specifically? There's a variety of GPU fixes for Hackintosh setups; maybe something from there can be implemented?