corna/me_cleaner #371: What possible bugs can be after removing ME?

necros2k7, 2022-01-08:
Any change in performance? Overclock capabilities? Fan / sleep mode broken?

duketwo, 2023-02-03:
For me it seems that ME is managing the BLCK at least, also usually the startup time (until post) takes 10-20 seconds longer than without the mod.

Espionage724, 2023-07-09:
If you use the DSP for onboard audio (Intel SST), firmware can't load for it on Linux: https://github.com/thesofproject/sof/issues/4923#issuecomment-1616912022

My computer wasn't using DSP by-default and I can toggle it relatively easily, but more locked-down firmware might be an issue.

https://thesofproject.github.io/latest/getting_started/intel_debug/introduction.html#firmware-binary

> The Intel ME (Management Engine) is responsible for authentication of the firmware, whether it is signed by an Intel production key (consumer products), a community key (open development systems and Chromebooks since Gemini Lake) or an OEM key. If the Intel ME is disabled by an OEM, or disabled by user-accessible BIOS options, the firmware authentication will fail and the firmware boot will not complete. If the ME is disabled by the OEM, the only solution is to fall-back to the legacy HDAudio driver. If the ME is disabled by the user, the user must re-enable it. Unfortunately, no documented mechanism exists for the Linux kernel to query whether or not the firmware authentication is enabled, which means dmesg logs cannot be provided to alert the user to an ME configuration issue.