corna/me_cleaner #373: anyone tried booting a i9 skylake-x ?

candicewithassburgers, 2022-02-03:
hey i am quite new to this stuff , so go easy on me

anyone tried booiting a i9 skylake-x with a motherboard that has intel me disabled ?
and even better has anyone tried booting a intel i9 7900 x or 7980XE ?

portellam, 2022-02-03:
Best to check other forks that are ahead of _corna/me_cleaner_. I have had success with the fork _dt-zero/me_cleaner_. I have a 9th gen i9, with the HAP bit set in my BIOS. No issues since I did this five months ago.
Hope you have success!

candicewithassburgers, 2022-02-04:
> Best to check other forks that are ahead of _corna/me_cleaner_, and that work. I have had success with the fork _dt-zero/me_cleaner_. I have a 9th gen i9, with the HAP bit set in my BIOS. No issues since I did this five months ago. Hope you have success!

did you install coreboot in the 9th gen motherboard ? if so whats the model ?

portellam, 2022-02-04:
> did you install coreboot in the 9th gen motherboard ? if so whats the model ?

> https://www.coreboot.org/Supported_Chipsets_and_Devices
> _This page was last edited on 27 March 2015, at 16:15._
No.

FrostKnight, 2022-02-06:
I tend to think, that without coreboot, there isn't much point, as it could get reset by intel if they so choose...

:/

portellam, 2022-02-07:
> I tend to think, that without coreboot, there isn't much point, as it could get reset by intel if they so choose...

If I read you correctly, you could argue whether or not having a LAN port connected to Internet, or using a proprietary (brand-name) CPU, motherboard, etc. of which you cannot personally audit, or even using an Internet browser or Internet in general.

Experience taught me to pull the brakes when I realize day-to-day activities are impeded by schizophrenia brought upon potential attack vectors or what-if's. I don't know everything and I don't care.

Don't worry about that. Have fun learning. Buy a Coreboot laptop if you want. More power to you.

FrostKnight, 2022-02-08:
> > I tend to think, that without coreboot, there isn't much point, as it could get reset by intel if they so choose...
> 
> If I read you correctly, Yes there is a point. Your system will stop Intel ME from phoning home, but it will still operate like the head of a headless chicken (if I am correct).
> 
> You could argue the security of having a LAN port connected to Internet, or using a proprietary (brand-name) CPU, motherboard, etc. (of which you cannot personally audit), or even using an Internet browser (or Internet in general). Experience has taught me you need to pull the brakes, when you realize your day-to-day activities are impeded by schizophrenia brought upon by imagined potential attack vectors or what-if's.
> 
> Don't worry. Have fun learning. Buy a Coreboot laptop if you want. More power to you.

It also has more speed then the original bios and microsoft's stupid hands can't taint it. ;)

Btw, I do use an ath9k wifi card, which is why I recommended coreboot + intel me disabled rather than just intel me disabled.

You are right to some extent, I do indeed have some fear, due to so many fascists trying to get into power around the world.

That being said, I also don't think corporations should get everything they want without a fight, or really anyone for that matter. This is for the reasons above.  

Also data collection centers, abuse water when they could just add cooling units, to not destroy the environment as much, but they don't because it is cheaper...

All this being said, github does falls into the last category, my point is more, starve them in a way that doesn't outright make your own life full of despair.

It is possible you are right though, regarding your own use case. Maybe it is possible to use a different wifi card if you want to, with intel me being the only thing disabled. I have no idea... if it doesn't matter, then no worries.

If none of this is important to you, or worth doing, then just have a nice day.

Just voicing my opinions for anyone here, in case they want to be aware of such things.

Peace then!

portellam, 2022-02-08:
Keep in mind, OP asked about me_cleaner, and I gave an answer. OP asked something derivative. I gave another answer, Coreboot is NOT supported by newer hardware. Coreboot was/is developed by one person. This discussion is best for another thread.