corna/me_cleaner #374: any known me_cleaner forks that works on a 8th gen Xeon cpu ?

candicewithassburgers, 2022-02-08:
any known me_cleaner forks that will work on intel xeon E-2288G ?
i known that dt-zero/me_cleaner has worked on a 9th gen i9 (dont know if its desktop or laptop)

portellam, 2022-02-08:
> any known me_cleaner forks that will work on intel xeon E-2288G ? i known that dt-zero/me_cleaner has worked on a 9th gen i9 (dont know if its desktop or laptop)

I understand why you want to make a new issue request, understand you are the focus, not others. Have you tried running dt-zero/me_cleaner on a Desktop 9th Gen i9, for the first time already? Please don't get ahead of yourself. As for
>desktop or laptop
I don't care to share my personal hardware information, I thought that would be obvious, but Yes I run on Desktop. I've had success with earlier gen Intel desktops with corna and dt-zero. I know E5-2000 v3 (?) series CPUs on Chinese X99 motherboards work with dt-zero. However, certain brands like HP or SuperMicro may NOT support BIOS modding. You can check Miyoconst's channel on YouTube for more on BIOS modding that platform. YMMV.

candicewithassburgers, 2022-02-12:
i cant use dtzero/me_cleaner on a 9th gen intel i9 because well , i dont have one
i am planning to buy a new set up soon that why i am asking anyone if it will work because i dont want to waste the money i have spent for a long time on another bloated backdoor infested hardware

the company 9elements managed to install coreboot on a supermicro x11sch-f with a intel xeon e2276g (just 1 level below 2288g)
https://review.coreboot.org/c/coreboot/+/37441
https://9esec.io/blog/next-generation-coreboot-server-platform/

i was just wondering if anyone has managed to disable intel-ME on an intel xeon
because from what i can find , the x11sch-f is the newest(and possibly only) motherboard that supports both coreboot and  both 9 and 8 th gen intel

the only alternative to this is going 7th gen because disabled me with propiatery firmware is useless since there are may be(and probably are) backdoors in firmware

portellam, 2022-02-14:
lol Good luck.

root-hardenedvault, 2022-02-15:
There are a few options and one is worth to try is that replace SPS with some CSME extracted by KabyLake/CoffeeLake fw, which can be neutralized by me_cleaner. We've done the similar things during writing the [report (What every CISO and security engineer should know about Intel CSME)](https://hardenedvault.net/2021/07/16/ciso-seceng_csme.html) last year.

frantic-from-paranoiawf, 2024-12-03:
> > any known me_cleaner forks that will work on intel xeon E-2288G ? i known that dt-zero/me_cleaner has worked on a 9th gen i9 (dont know if its desktop or laptop)
> 
> I understand why you want to make a new issue request, understand you are the focus, not others, and the issue should stay on topic. Have you tried running dt-zero/me_cleaner on a Desktop 9th Gen i9, for the first time already? Please don't get ahead of yourself. As for
> 
> > desktop or laptop
> 
> I just don't care to share my personal hardware information, I thought that would be obvious. Yes I run on Desktop. I've had success with earlier gen Intel desktops with corna and dt-zero.
> 
> I know E5-2000 v3 (?) series CPUs on Chinese X99 motherboards work with dt-zero. However, certain brands like HP or SuperMicro may NOT support BIOS modding. You can check Miyconst's channel on YouTube for more on BIOS modding that platform. YMMV.

He was obviously saying he knows others have gotten it to work on that processor.

frantic-from-paranoiawf, 2024-12-03:
> lol Good luck.

Don't comment on an issue if you have nothing to say. Don't be a useless cock, he's asking everyone, not you specifically.

portellam, 2024-12-04:
@frantic-from-paranoiawf 
Nice necropost.

> 
> Don't comment on an issue if you have nothing to say. Don't be a useless cock, he's asking everyone, not you specifically.
>
Opening a new thread when one exists already adds noise. Consider how quiet this repo and this is not a forum, objectively it doesn't matter and I grant you that.

From what I recall, I was helpful and tried to stave off the OP's all-or-nothing outlook. You have to check out other forks for newer hardware support (see previous issue #373, or [dt-zero's fork](https://github.com/dt-zero/).

>
> He was obviously saying he knows others have gotten it to work on that processor.
>
Yeah, me (see #373). Nice false argument. I provided help and _when I felt that my_ advice was ignored, I was frustrated. What did I learn? Be more patient and not add noise.

frantic-from-paranoiawf, 2024-12-05:
> @frantic-from-paranoiawf Nice necropost. Let me ask you, are you helping?
> 
> > Don't comment on an issue if you have nothing to say. Don't be a useless cock, he's asking everyone, not you specifically.
> 
> Nice false argument.
> 
> From what I recall, I was helpful and tried to stave off the OP's all-or-nothing outlook. Trust me, I've been there; I wasted the Summer of 2019 trying to do all the things: coreboot, me_cleaner, and VFIO a PC, I could not get coreboot to work.
> 
> You have to check out other forks for newer hardware support (see previous issue #373, or [dt-zero's fork](https://github.com/dt-zero/).
> 
> Opening a new thread when one exists already adds noise. Consider how quiet this repo and this is not a forum, objectively it doesn't matter.
> 
> > He was obviously saying he knows others have gotten it to work on that processor.
> 
> Yeah - me. (see #373) I provided help and _when I felt that my_ advice was ignored, I was frustrated. What did I learn? Be more patient and not add noise.

No, I am not helping. Doesn't stop me from telling you that you aren't. I am happy to admit I'm not helping.
It's a dead repo and issue, who cares. You were commenting when the issue was alive. If the repo was still alive, it might actually be a good idea to necropost, since it would alert the repo owners to close the issue, but that's not happening in this case.

Seems like a skill issue if you ask me. As long as the codename of your CPU is supported, blobs can easily be downloaded from vendor or be redistributed, and your chipset is supported, you can probably port any board without much effort. I mean, I was able to make a port for an Intel 10th gen desktop motherboard on my first go. I finished the port, but realized FSP for this chipset was NDA-only. So I couldn't publish it.

The forks you posted in your reply are as dead as upstream. You need to simply fork your own and push the existing pull requests that add support for newer hardware.

If you ask me, I don't understand how you got annoyed. He's asking a reasonable question.
You answered him saying you successfully did it with a 9th generation i9, and you don't have coreboot. He didn't speak again in that issue.
He opened another issue asking about Xeon instead. You linked him to entirely different Xeons.
coreboot and ME cleaning can both be finnicky, so it's totally reasonable to ask whether a specific CPU is supported or not.
It's like a knife to the heart if you buy hardware with a requirement in mind, but then it just doesn't work.