corna/me_cleaner #378: me_cleaner results on ThinkPad T440?

ThinkerPadder, 2022-04-16:
Hello,

I know that me_cleaner works well with the ThinkPad T440 **P** and possibly the T440 **S**, but I am not sure about the standard T440.

If anyone knows anything about T440 me_cleaner compatibility, your knowledge would be greatly appreciated.

By the way, is there such thing as Coreboot or a Coreboot equivalent for the T440?

Thank you for any information!

digmorepaka, 2022-04-18:
T440 uses the same version of iME as T440s, same parameters apply. 

There is no path to coreboot for a T440 unless one of these requirements is met:

ES/QS/bootguardless retail CPU is soldered

You find a bootguard bypass

ThinkerPadder, 2022-04-19:
I'm not too familiar with this advanced terminology (such as ES, QS), could you please point me to somewhere where I could learn more about these things?

So forgetting Coreboot, you're saying that the ME cleaner would work the same on a T440 as on a T440s?

Assuming that is certainly the case, now the question is, how to flash ME cleaner to a T440 BIOS chip without desoldering? I've never done this sort of thing before, I'm not even sure which pins to connect to on a Raspberry Pi.

Thank you for your answer, it is much appreciated.

czeej, 2022-04-19:
I have a T440, it works.. Partially. Not good for the chip set but. more then Skylake.

ThinkerPadder, 2022-04-19:
@czeej could you please kindly simplify that and format it into steps for the sake of me, a noob and complete beginner regarding me_cleaner, better understanding this?

I hope that is not too much to ask, sorry for my inexperience!

Here are some questions regarding what I did understand:

1. May I see the video that you mentioned?
2. The breadboard suggests that I will have to de-solder my BIOS chip. Is that the case?
3. '3.1v power supply or fix it.' What do you mean by 'fix it'?

Thank you!

czeej, 2022-04-19:
You shouldn't need to desolder if you use a soip 8 clip. You attach the wires to the bread board then the clip to the board. 
watch?v=aRUxfxp9dJ8  (there is video on youtube) I'd just download it with youtube-dl and keep it for reference. Just in case it ever disappears.
The CH341a programmer uses a 5v rail to supply power which is not recommended.
Simply dump two back ups.  Diff the dumps to make sure your reads are good. 
Run me_cleaner, 
flash the modified image.
See the me_cleaner status issue board (the first page first post). I have a post regarding the T440 and what works.

ThinkerPadder, 2022-04-20:
@czeej I like how he's making a video about removing backdoors and proceeds to boot into one of the most 'Alphabet Boys' (quoting Mental Outlaw) compromised operating systems in existence, Windows 10. Haha.

/

Now for some questions (again, please forgive me for my noobiness -_-):

01. 

I've heard the term 'dumping' alot when researching me_cleaner, Coreboot etc: What exactly does 'dumping' (file dumping?) do in this case? I assume that the files dumped to one of the PC's drives, another PC/Ras Pi where the CH341a is attached, or to a chip on the motherboard? I'm not sure. 

Is this all automated or do I have to pick a manual dumping 'point'?

02.

Does this 3.1v PSU come with the CH341a or does it need to be purchased separately? I found some reference to 3.3v in my research but nothing relating to 3.1v.

/

btw, I notice now that the chip doesn't need to be desoldered since it doesn't have to be the one on the breadboard, the CH341a can take that position: I was confused with that earlier.

I saw your status issue board post :)

Many thanks again for your assistance! If it weren't for you, I would have probably given up.

czeej, 2022-04-20:
1. flashrom -p ch341a -r fulldump.bin  by dump it just means reading the flash and storing it outside the flash. Dumping the flash contents onto your hard drive.
2.  you are right 3.3v  see this site [https://libreboot.org/docs/install/spi.html](url)
3. you can use the ch341a and make things easier but reading the above it's not 'correct' voltage. Be very careful not to short anything out.

Remember to make two dumps before writing anything! diff the two and make sure the clip and everything is working properly.
diff fulldump1.bin fulldump2.bin if no output they are the same. Then verify the image with me_cleaner.py -c fulldump.bin
Make sure it's not a corrupted read. 
Then run me_cleaner ./me_cleaner.py -S -w (with your two partitions you need) -O cleaned.bin 
then flashrom -p ch341a -w cleaned.bin

ThinkerPadder, 2022-04-20:
Thanks for all this insight! I think that's all I will need to know for now. I'll let you know of my progress.

btw, I am curious about the custom BIOS you mentioned on the status issue board. Please kindly let me know of any progress on that. 😃