corna/me_cleaner #381: Intel AdlerLake

phodina, 2022-05-30:
Hi,

I've attempted to run the `me_cleaner.py` on 12th Intel Core i7-12700 on MSI PRO Z690 and the `UEFITool` shows the version of the `ME Region` is  `16.0.151545`.

Is there a guide on how to read the region and remove the ME?

Would it be helpful to link the dump of the ROM?

renehoj, 2022-08-02:
Here is a dump of the Dasharo Z690 rom, I case it helps anyone find the HAP bit

`File dump.rom is 33554432 bytes
ICH Revision: Unknown PCH
FLMAP0:    0x00040003
  NR:      0
  FRBA:    0x40
  NC:      1
  FCBA:    0x30
FLMAP1:    0x73100208
  ISL:     0x73
  FPSBA:   0x100
  NM:      2
  FMBA:    0x80
FLMAP2:    0x00140170
  PSL:     0x1401
  FMSBA:   0x700
FLUMAP1:   0xc0ff02df
  Intel ME VSCC Table Length (VTL):        2
  Intel ME VSCC Table Base Address (VTBA): 0x000df0

ME VSCC table:
  JID0:  0x0000471f
    SPI Component Vendor ID:            0x1f
    SPI Component Device ID 0:          0x47
    SPI Component Device ID 1:          0x00
  VSCC0: 0x20152015
    Lower Erase Opcode:                 0x20
    Lower Write Enable on Write Status: 0x06
    Lower Write Status Required:        No
    Lower Write Granularity:            64 bytes
    Lower Block / Sector Erase Size:    4KB
    Upper Erase Opcode:                 0x20
    Upper Write Enable on Write Status: 0x06
    Upper Write Status Required:        No
    Upper Write Granularity:            64 bytes
    Upper Block / Sector Erase Size:    4KB

OEM Section:
00: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
10: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
20: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
30: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff

Found Region Section
FLREG0:    0x00000000
  Flash Region 0 (Flash Descriptor): 00000000 - 00000fff 
FLREG1:    0x1fff1000
  Flash Region 1 (BIOS): 00000000 - 00ffffff 
FLREG2:    0x03d80001
  Flash Region 2 (Intel ME): 00001000 - 003d8fff 
FLREG3:    0x00007fff
  Flash Region 3 (GbE): 00fff000 - 00000fff (unused)
FLREG4:    0x00007fff
  Flash Region 4 (Platform Data): 00fff000 - 00000fff (unused)

Found Component Section
FLCOMP     0x0930f0f6
  Dual Output Fast Read Support:       not supported
  Read ID/Read Status Clock Frequency: 33MHz
  Write/Erase Clock Frequency:         33MHz
  Fast Read Clock Frequency:           33MHz
  Fast Read Support:                   supported
  Read Clock Frequency:                20MHz
  Component 2 Density:                 32MB
  Component 1 Density:                 32MB
FLILL      0x00000000
  Invalid Instruction 3: 0x00
  Invalid Instruction 2: 0x00
  Invalid Instruction 1: 0x00
[... 147 lines cut ...]
  Requester ID:                      0xffff

FLMSTR3:   0xffffffff (GbE)
  Platform Data Region Write Access: enabled
  GbE Region Write Access:           enabled
  Intel ME Region Write Access:      enabled
  Host CPU/BIOS Region Write Access: enabled
  Flash Descriptor Write Access:     enabled
  Platform Data Region Read Access:  enabled
  GbE Region Read Access:            enabled
  Intel ME Region Read Access:       enabled
  Host CPU/BIOS Region Read Access:  enabled
  Flash Descriptor Read Access:      enabled
  Requester ID:                      0xffff

Found Processor Strap Section
????:      0xffffffff
????:      0xffffffff
????:      0xffffffff
????:      0xffffffff
????:      0xffffffff
????:      0xffffffff
????:      0xffffffff
????:      0xffffffff
`

renehoj, 2022-08-06:
I don't know if it works with the MSI firmware, https://github.com/Dasharo/dasharo-issues/issues/111

phodina, 2022-11-08:
Well not sure what you mean. I flashed the Dasharo Coreboot so their is no longer any proprietary MSI firmware.

Also the code for ME is in the SPI Flash in special "section" and should be independent of the Bios.

The ME itself loads the firmware upon boot, correct?