corna/me_cleaner #383: Easy way to disable ME on Samsung laptops

disableme1, 2022-07-15:
This method doesn't require any firmware modification or hardware access, disables the management engine on subsequent reboots until a shut down. So there's no risk of bricking the hardware and it's easy to revert. One drawback is, while disabled the ME region will be unlocked for read/write access which could pose a risk in case an advanced malware or adversary gains access to your system.

There is a hidden BIOS menu on (some?) Samsung laptops that gives access to advanced settings. It appears on the Exit section of the BIOS screen after pressing these keys at the same time: Ctrl + Alt + Shift + F4

There, under ME settings is an option to turn off sending the "End of POST" message to the management engine. Switching this off allows you to send certain commands to the ME interface from your OS to temporarily disable it.

This document explains the commands that disable ME under various conditions:

https://github.com/ptresearch/me-disablement/blob/master/How%20to%20become%20the%20sole%20owner%20of%20your%20PC.pdf

I first tried the third method (Soft temporary disable) but it didn't do anything. Then I tried the second one: HMR FPO - Host ME Region Flash Protection Override. On next reboot the management engine was turned off. It is supposed to work for just one reboot but it still stays off on further reboots, so either the document is inaccurate or the disabled "End of POST" message is helping keep the ME turned off.

To send the commands, I use the me_util.py tool from here: https://github.com/skochinsky/me-tools

The script works with 32-bit Python 2.7 on Windows, it can probably be turned into an .exe file for easier use.

I use this command for the second method (HMRFPO) from the document. It will only work if you already disabled the "End of POST" message from the hidden BIOS settings.

`python me_util.py 0x05 0x01 0000000000000000`

Using this command and rebooting (not shutting down) the laptop will disable ME. After any shut down, just use the command again and reboot to disable the ME again.

0x199x, 2022-12-26:
Interesting, and thanks for sharing, but maybe important to note that this is for Windows only (me_util.py needs access to ME drivers).  If it was a one-off procedure, it wouldn't be such a big deal, but since the command must be rerun after every shutdown, it basically requires that Windows be available at all times.

doritos4mlady, 2024-02-11:
Cool. Which models have the hidden bios screen?