corna/me_cleaner #389: Toshiba C850-1G2 - 30 minute timer.

yotoprules, 2023-01-10:
Hi,
I have a Toshiba C850-1G2 that came with a Celeron 1000m Ivy Bridge. I want to put in an i3 2350m sandy bridge.
I did some research and apparently the 30 minute timer is due to the ME, and I was told the 30 minute timer should go away if I use this tool.
I was able to modify the BIOS and flash it just fine - I also backed it up and did "-c" to verify it and it says that the MeAltDisable bit is enabled.

However, despite this, the 30 minute timer is still in effect. at exactly 30 mintues the system just shuts down.

Am I doing something wrong? I did flash it VIA InsydeFlash which appeared to flash it no problem.

I did initially try to use CH341a but for some reason, this motherboard has 2 BIOS chips, which are both each 2MB, which totals 4MB, but the full BIOS file is 6MB, so I have no idea how I would go about flashing the modified BIOS this way - I did make a dump of both chips anyway.

Any ideas?

Thanks!

cuvtixo, 2023-01-11:
I'm afraid you have a fundamental misunderstanding here. Or a fundamental miscommunication. Celeron and i3 are the names of the chips themselves. "Ivy Bridge" and "Sandy Bridge" are, as the name implies, serve as the "bridge" between the CPU and the rest of the components of the motherboard. Only CPUs made for Ivy Bridge will work on Ivy Bridge motherboards. Only Sandy Bridge CPUs will work on Sandy Bridge motherboards.  If you can fit an entirely different motherboard, a Sandy Bridge, in the Toshiba C850-1G2 (extremely doubtful) you could use this new CPU. Otherwise, no. The BIOS has nothing to do with it (well, very little). Reflashing won't help. 
I admire your ambition, but there is no way this is ever going to work.

yotoprules, 2023-01-11:
Hi, thank you for the response.
I know what ivy and sandy bridge are.
Many sandy and ivy bridge laptops where with both gens of CPUs: I upgraded a Lenovo from a i3 2328m to a i5 ivy bridge (forgot the model but it was a 3000 series).
The problem here is that Intel artificially put in a limitation in the HM70 chipset which sets a 30 minute timer if you put any i3,i5,i7 in a HM70 board that would've shipped with a celeron or pentium.the i3,i5, and i7s would've shipped.with a HM76 board.
I have done some more research, and it seems that the 30 minute timer is coded into the chipset PCH, not the firmware or ME. Which means I wasted hours of my time (my fault). I was able to successfully clean the ME though.
Unfortunately the people who said cleaning the ME would remove this specific timer were incorrect, as I have done research and found that no-one has been able to remove this timer without replacing the HM70 chipset with a HM76 chipset, which is well beyond my skill level and not worth it for this budget laptop with the worst speaker I've ever heard.

TLDR for anyone finding this: the timer is coded into the Chipset, and cannot be removed. Cleaning the ME won't fix it. Without replacing the chipset or the motherboard, you will be stuck with a celeron or pentium if you don't want the timer.