corna/me_cleaner #390: I may have found a MODERN laptop with INTEL ME corrupted but refusing to turn off after 30 minutes.

Hunterrules0-0, 2023-02-03:
Whenever I boot the laptop its able to start up however it seems that Intel me is disabled this could be a breakthrough in finding a way to disable it. This laptop was a replacement for my old one that I got from a repair shop. It has not ran me_cleaner and is refurbished

Turning the laptop on and plugging in my pop os usb I get treated to [This warning message](https://www.youtube.com/watch?v=j1nFVWoExiI)

It stays here for about 3 seconds before continuing the normal boot 
After that everything works just as intended. 

Entering the intel management setup screen(Yes this laptop has that) I get treated to [This error](https://www.youtube.com/watch?v=vM3mxXQ_tPQ) Which once is finished kicks me out into the normal booting process.

It appears that the intel mangement engine firmware is corrupted and somehow isnt shutting off after 30 minutes
along with that saving and exiting the setup gives me this werid Beep code(https://youtu.be/iNR3we6OtmE) thats unlike any beep code ive ever seen. it could be a beep code for intel me. after the beep code it refuses to restart and goes into a black screen I have to force restart the computer to which finally the computer starts normally. However restarting it reveals that the changes ive made in the setup have been successfully saved. Which leads me to conclude further that the intel mangement engine is corrupted

perhaps theres a way you guys could use this to help you find a way in permentantly disabling the management engine if you have any questions fell free to ask
Have an amazing day

Hunterrules0-0, 2023-02-05:
EDIT: I was able to dump the me sector of the bios with flashrom
[image]
Dumping this part of the mangement engine should not work Me prevents this from happening however here it just straight up gives me the me region

Hunterrules0-0, 2023-02-10:
edit: I forgot to dump the whole bios so sorry heres the whole thing

https://gofile.io/d/t0Cejm

goryramsy, 2023-02-13:
I'm just subscribed to this repo, mainly for interesting ME developments like this, so I am in no way qualified to answer questions here. This does look odd to me too.

Mind sharing the laptop version? It says Lenovo, but which Lenovo laptop specifically?

ValoWaking, 2023-02-14:
ME is only end user interface to receive info from sensors and make / save some bios preset in bios.

this does not mean that there is no backdoor in the device. This indicates the laziness of the manufacturer who released the bios for this device.

CodeAsm, 2023-02-14:
What exactly makes you believe Intel ME is not working?

If I read the error message in both videos, the Bios firmware could not send its "end of POST" to the Intel ME over the "HECI" communication channel. [DCMI-HI
DCMI Host Interface Specification](https://www.intel.com/content/dam/www/public/us/en/documents/technical-specifications/dcmi-hi-1-0-spec.pdf ) is where i found more information about this HECI communication. especially chapter 3.1.

Im in no way experienced with Intel ME, and particular in your system, I have no idea what is actually happening or in what state, Intel ME is in. There could be more wrong, or even a hardware issue. It seems "only" this communication channel isnt working, the same documentation (in the introductory part) explains this is not the only path for ME and the HOST to talk to eachother, this may mean Intel ME is still fully operational and functional. It might even mean when the HOST boots a OS, Intel ME is still able to perform its duties and communicate with say modules or drivers inside the OS.

Id be intresting to know what the actual detected Intel ME status would be: https://github.com/corna/me_cleaner/wiki/Get-the-status-of-Intel-ME

But im just a fellow user, no experience and thus have to warn you not to just run code from unkown sources. Great you dumped the bios :D and hope your OS of choice is working as intended. if not, try updating the bios or installing chipset specific drivers. yes, this might "fix" the current fault.

CodeAsm, 2023-02-14:
Also, you probably broke the bios normal boot procedure (inpatient reboot, when you deselect certain bios settings from being enabled to disabled, the bios needs to fully reboot the whole system instead of a "soft" reboot).

you may have semi bricked the system doing so. restoring the bios variable storage might fix this. 

On your youtube video I also posted this:
`
1. Remove all power resources (the ac power adapter, removable battery, and coin-cell battery). If your computer has a built-in battery, reset the computer by inserting a straightened paper clip into the emergency-reset hole. Wait for one minute. Then reconnect all power resources.
2. Replace the system board (service provider only)
`

I got this specific error because I let the Lenovo diagnositic app detect the error from your video. get the app and more information from here: [https://support.lenovo.com/us/en/solutions/ht503581-lenovo-pc-diagnostics-using-beep-codes-thinkpad](https://support.lenovo.com/us/en/solutions/ht503581-lenovo-pc-diagnostics-using-beep-codes-thinkpad)

I hope you have a way to recover the board to a working state.

Hunterrules0-0, 2023-02-15:
Wow was not expecting so many replies. Anyways I belive the me is corrupted as normally your not allowed to dump the me region with flashrom however with this machine it just straight up dumps the entire thing with the internal programmer
second no matter what I do its impossiable to enter the intel mangement setup menu in f12 doing so gives me those error codes
Also the machine is able to boot perfectly fine(im even typing this on the laptop). As soon as I reset the laptop after those post codes in the video I showed the laptop will perfectly start up again with no problems(even the changes in the setup are saved)

Hunterrules0-0, 2023-02-15:
> I'm just subscribed to this repo, mainly for interesting ME developments like this, so I am in no way qualified to answer questions here. This does look odd to me too.
> 
> Mind sharing the laptop version? It says Lenovo, but which Lenovo laptop specifically?

The laptop is a leveno t470. Ill give you more info later I promise 

have a good day

Hunterrules0-0, 2023-02-15:
EDIT I WAS RIGHT ME IS CORRUPTED. I ran the metool and it came back with this

[image]

CodeAsm, 2023-02-16:
from what I understand, yes, looks like a broken Intel ME image or atleast communication to it in some ways. possibly other ways are still working. If Intel ME is still happily running, we dont know? altho intelmetool leads me to believe its still running, just crippled a bit? 

does ``lspci | egrep '(ME|HECI)'``
still result in being found?

Hunterrules0-0, 2023-02-16:
> from what I understand, yes, looks like a broken Intel ME image or atleast communication to it in some ways. possibly other ways are still working. If Intel ME is still happily running, we dont [...]

It seems so yeah sadly perhaps it is still running
[image]

Hunterrules0-0, 2023-02-16:
for some reason it says sunrise lake. despite the fact that this machine is skylake

Hunterrules0-0, 2023-02-16:
EDIT: HUGE UPDATE
as it turns out. it seems that ME will not appear when using lspci if you boot from the f12 menu. for some reason booting from the disk in the f12 menu makes intel me vanish.  this appears to only work by booting the disk in the f12 menu 
[image]

CodeAsm, 2023-02-16:
I dont emediatly understand what the difference would be between normal boot and F12 (I know, its some efi boot menu thing) but maybe its not exposing the pci device correctly now? I might be able to look but im also afraid im hunting red herrings.

What I find odd, but maybe this can happen, is your video shows a bios version 1.72 and the ECP seems to be 1.33
```
  Package      (ID)     UEFI BIOS (BIOS ID)  ECP (ECP ID)      Rev.  Issue Date
  --------------------  -------------------  ---------------   ----  ----------
  1.73      (N1QUJ39W)  1.73 (N1QET98W)      1.36 (N1QHT54W)   01    2023/02/02
  1.72      (N1QUJ38W)  1.72 (N1QET97W)      1.36 (N1QHT54W)   01    2022/11/02
  1.71      (N1QUJ37W)  1.71 (N1QET96W)      1.36 (N1QHT54W)   01    2022/08/24
[snip]...
  1.56      (N1QUJ22W)  1.56 (N1QET81W)      1.34 (N1QHT52W)   01    2019/03/04
  1.55      (N1QUJ21W)  1.55 (N1QET80W)      1.33 (N1QHT51W)   01    2019/01/29
```  
 Maybe the refurb company did flash the (then) latest bios, but dint use normal update methods? did you by any chance did a full reset to defaults in the bios any time? I also think that running any normal update procedure might either get rid of this situation and "fix" your bios, or eh... brick it. but I think you only have a broken coms line to the ME part of your cpu

Hunterrules0-0, 2023-02-16:
> I dont emediatly understand what the difference would be between normal boot and F12 (I know, its some efi boot menu thing) but maybe its not exposing the pci device correctly now? I might be able [...]

Personally having the intel mangement engine not working is a win for me however I still believe that more than just the coms line is broken. As that wouldnt explain why the computer was able to access and dump the me section of the bios without  issue which is normally impossible with flashrom as its usally locked and cannot be dumped without an external programmer i believe however with this laptop it does it no question

Hunterrules0-0, 2023-02-16:
Edit dumping the bios again with the option -p internal:this_is_not_a_laptop

frantic-from-paranoiawf, 2024-12-03:
Judging by some of your screenshots, even though it reports that the implementation is "broken", it still is technically running in it's normal phase (not disabled).
The missing PCI thing I can't exactly explain myself. But I'd have to guess ME was still running.
Were you able to check if BootGuard was enabled on this specific laptop? Because if it was disabled, then the previous owner probably did something to the ME or firmware. I know some models were privately made with no fused keys, they most likely could've also just did BGA work and put a new unfused PCH on the motherboard.

frantic-from-paranoiawf, 2024-12-03:
Also note, you can disable ME after BringUp on multiple upon multiple modern laptops that are widely available. I'm talking like Alder Lake modern.