corna/me_cleaner #392: how to shrink 11mb (from 16mb chip) hp8300 into smaller ME but keep it active because need for hd4000 osx acceleration? thx

kocoman1, 2023-02-07:
how to shrink 11mb (from 16mb chip) hp8300 into smaller ME but keep it active because need for hd4000 osx acceleration? thx

ValoWaking, 2023-02-14:
don't break your device. The ME operating system resides in non-rewritable memory in u chipset or CPU. Editing the BIOS firmware does not affect the operation of the backdoor in any way.

if u want short answer what is doing me_cleaner with backdoor - this is it - nothing.
but u can try broke SPI firmware on old platform like 945gc - nouthbridge have'nt ME, but ich7 - is a ready AMT southbridge 

and u all guys is doing nothing here many years.... how long will this continue?

d-wid, 2023-05-09:
> The ME operating system resides in non-rewritable memory in u chipset or CPU. Editing the BIOS firmware does not affect the operation of the backdoor in any way.

If so why is there an ME partition in the BIOS firmware, and why do various vendors provide updates to the ME software via BIOS updates?

d-wid, 2023-05-09:
@kocoman1 If I understand correctly I think me_cleaner.py -S -r -t -d -O $OUTPUTFILE $INPUTFILE (the input file being a HP8300 BIOS image) is what you're looking for.