corna/me_cleaner #399: What is the reason that has stopped the development of this project?

collector-ynh, 2023-07-08:
Hello, I would like to know why this project has been blocked. Were the developers pressured by certain entities?

Espionage724, 2023-07-13:
https://www.youtube.com/watch?v=7gRsgkdfYJ8

I don't think you're going to get a direct answer :p If I recall right, the official dev response was not enough time or other priorities.

I could be wrong, but my understanding was that me_cleaner came out before HAP bit was discovered. HAP bit seems like the most ideal method for disabling ME, and I thought there was a version of ME that started a trend where removing or altering partitions either didn't work or wasn't viable from that version of ME and some or all versions after that. I'm not aware if the newest versions of ME can have the HAP bit set or if anyone has tried to, but it seems so long as the HAP bit works there isn't much development needed.

I'm on Coffee Lake and I think I had me_cleaner from a fork, but I recall it saying something like partitions can't be removed. HAP bit seemingly works fine and ME gets disabled.

collector-ynh, 2023-07-13:
Thanks for your reply, but some people consider the HAP BIT insufficient, otherwise Intel would have already found a solution to remove it from new versions!

Tu parles français ?

corna, 2023-07-26:
Hi,

as @Espionage724 said, mainly lack of time and interest. Tinkering with the ME binary images take **a lot** of time, as the development continuously bricks the test machine, requiring time-consuming programmings through an external programmer. I had a lot of spare time in the past (and a lot of interest in disabling ME, since at the beginning there was no way at all), this has now changed.

Hopefully in the future I will resume the work (and check the accumulating issues and PR).

> Were the developers pressured by certain entities?

No, I have never been contacted by Intel or anyone else (except before my [34C3 presentation](https://media.ccc.de/v/34c3-8782-intel_me_myths_and_reality), they just wanted to know the content of the presentation), I suppose they do not care very much.

> I could be wrong, but my understanding was that me_cleaner came out before HAP bit was discovered. 

Yes, roughly 1 year before.

> HAP bit seems like the most ideal method for disabling ME

Yes and no. It is certainly "cleaner", but actively removing the unneeded code may be considered preferable.

closed 2023-07-26 by corna

collector-ynh, 2023-07-30:
@corna Thank you for your reply. I really hope you'll be able to resume the great work you started with versions higher than ME 11. 

Coreboot is actively using your tool and some people don't trust HAP-BIT and prefer effective ME reduction!

hanetzer, 2023-07-30:
>some people don't trust HAP-BIT and prefer effective ME reduction!

plus, removed modules means more space for coreboot and payloads :)