corna/me_cleaner #401: Re-enable ME or reset HAP bit without hardware re-flashing? (CFL)

Espionage724, 2023-08-09:
I dumped the BIOS chip off a laptop, ran me_cleaner for the HAP bit and descriptor removal, and re-flashed it back. ME is in Alt disable mode. I'm interested in re-enabling ME back to factory defaults, and would like to avoid having to physically re-flash the old image.

I thought a Dell BIOS update would do it but it seems to flash everything else and fails with ME. I have Coffee Lake.

LostOnTheLine, 2024-03-28:
I'm interested in this as well. I got an Optiplex 7050 Micro (or whatever Dell calls their 1Liter PCs) & discovered after trying extensively to access ME that the ME is disabled from the factory. I personally _**want**_ the ME on this machine but it seems according to Dell `"Once it's been disabled it is not possible to re-enable it"` so I'm looking for a way to do it anyway.

What you are looking for seems similar. Dell apparently intentionally doesn't allow it to be returned via the BIOS flash so it may likely require physically reflashing. If you have a guide or other information on how to do this I'll happily try it out & let you know how it goes

St1ckyNew, 2025-05-29:
> I dumped the BIOS chip off a laptop, ran me_cleaner for the HAP bit and descriptor removal, and re-flashed it back. ME is in Alt disable mode. I'm interested in re-enabling ME back to factory defaults, and would like to avoid having to physically re-flash the old image.
> 
> I thought a Dell BIOS update would do it but it seems to flash everything else and fails with ME. I have Coffee Lake.
> 
> Edit: Re-flashing with a clean BIOS image from Dell with USB recovery (BIOS.rcv) even keeps ME disabled

No Hardware needed to reflash dump or modified. ( Works on ASUS cant tell about the others, you could try )
[https://github.com/corna/me_cleaner/issues/402#issuecomment-2919176226](Instructions)