corna/me_cleaner #402: [HELP] Disabling IME on Coffe Lake

Morasithil, 2023-08-20:
I've just been reading alot instructions how to disable the IME and it's overwhelming, i must admit.. it's complicated.
I know only one thing certainly, i value privacy and freedom and i want to disable this BS IME on my system.
After all i'll quote Nicola Corna here:
"Even if it sounds dangerous, once you have a valid backup of your ROM and a way to reprogram it (external flasher, dual BIOS...), you should be safe."
So i would be a fool not to try.

My hardware:
CPU - INTEL Core i7 8700k
Mainboard - Asus ROG Maximus XI Hero Gaming 1151 ATX Z390
OS - Windows 10 Pro 64Bit V21H2
I've got a second computer with the same OS (no linux, but could install in a vm if required, and it's seems like it's required)
Would mint or debian be fine?
I have questions . . .

Correct me if wrong, basically what i'll have to do is edit a copy of the original bios firmware, by edit i mean "disable" ime, then flash the modified biosfirmware on the rom-chip / bios-chip?

From what i've seen, external flashing is highly recommended as internal flashing has limitations or might not work at all.
Here https://github.com/corna/me_cleaner/wiki/External-flashing a linux board is mentioned as requirement, i don't have that.
However if i understand correctly, alternatively to a linuxboard a USB SPI programmer would be another option.
You listed these variations: CH341A or the FT232H/FT2232H/FT4232H **(Guess i'll just pick the first one here?)**
Then you say this: "replace the -p linux_spi,... option in flashrom with the corresponding driver."
I don't understand how this process works, i know you want me to replace the above code with another one, but the how is missing for me.
What exactly am i supposed to do here https://wiki.flashrom.org/Supported_hardware#USB_Devices ?
Which of those: SOIC clip, DIP clip, SMD clips should i get?
When you say "Turn off the PC and disconnect it from the power supply" is it enough to disconnect the powersupply from any voltage source or did you precisely mean to disconnect all cable from the PSU that are connected to any pc components?
Removing the board-battery shouldn't be an issue.
This is the filename of the bios for my board DRV_Chipset_Intel_TP_W11_64_V101188368283_20211022R.zip
Does that seem right to you? Just want to be sure.
PS: I don't want to update the bios version because downgarding will not be possible anymore, screw asus!
If also been reading this guide https://github.com/corna/me_cleaner/issues/98
As i said before, this is overwhelming and somewhat complicated / confusing to me.

Morasithil, 2023-08-20:
Here a screenshot of the manual for the Asus ROG Maximus XI Hero Gaming 1151 ATX Z390 mainboard.

[image]

Espionage724, 2024-03-20:
The kind of chip clip you want depends on what your motherboard has. I would physically look at it as some motherboards can ship with different ones. I've only ever seen 8-pin BIOS chips and used a SOIC8 clip.

The BIOS file itself isn't specific to anything and can be named whatever, but typically you dump the BIOS image from the BIOS chip in something like "BIOS.bin", pass it through me_cleaner to output a modified file like "BIOS-mecleaned.bin", and then write that BIOS back to the BIOS chip.

I used a SOIC8 clip and a Raspberry Pi. I've done soft-disable-only to do the HAP bit and kept partitions in-tact on Coffee Lake (Dell Latitude 5591) and ME is disabled enough to not show a HECI device, and it even remains off on BIOS updates.

I imagine with a high-end desktop motherboard you may be able to flash to the BIOS chip through software but you might have to toggle settings in the BIOS to unlock it or set motherboard jumpers; might even be lucky enough to just flash the BIOS with USB flashback or something and could use a downloaded BIOS image. If the BIOS chip can be removed, you could also flash it from an older motherboard that's unlocked (I did that with a Skylake desktop's chip and old Phenom II motherboard).

The idea is that you need to be able to get a hold of the BIOS chip's contents, pass it through me_cleaner to modify it, and then re-flash it, the hardest part being re-flashing the modified image as most BIOS nowadays come with firmware locks. High-end desktop motherboards usually let you write. Laptops and pre-built desktops from OEMs usually don't, hence the SOIC/bios chip clip and going at the BIOS chip directly.