corna/me_cleaner #419: Dell Latitude 5591 Preserves HAP Bit through BIOS Updates/Recovery/Resets

Espionage724, 2024-09-08:
With BIOS updates and USB recoveries and CMOS resets, the HAP bit is seemingly preserved!

It interestingly passes Dell's trusted firmware checker too even though the platform hardware's firmware was technically modified outside the factory :p

I've been running with ME disabled on this laptop for a while now Linux, Windows 10/11, and recently FreeBSD 14.1 all without any noticeable issue.

MonterLuis, 2024-10-14:
Hello, I have read about how to disable the ME, but I am still worried about making a mistake, as I had just seen the Intel's processor of your laptop is gen8, right? Either a i7 8850H, an i5 8400H or 8300H  (coffe lake, both 3). Mine is an i7 8665U, whiskey lake. May I ask the exactly steps you follow to achive that on your laptop? Mine is a thinkpad x1 yoga gen4. I don't know the all the precautions I have to take to not destroy. I have so many questions, did you unsolder the bios flash chip? I plan to buy an exactly equal chip to load the bios and then exchange it to have the original as a backup. I appreciate any help.

Espionage724, 2024-10-20:
I have a picture: https://www.facebook.com/share/p/PydSrvzDugHZ2D89/

I used a SOIC clip on the BIOS chip while it was on the motherboard. Kind-of tricky with it being WSON but the SOIC clip I used had curved pins that happened to meet the pins while still being able to clip onto the WSON chip!

I didn't de-solder the chip mainly because I don't have a soldering station or any clean way of trying to do that safely :p But that would likely be far easier than trying to get a SOIC clip onto WSON! I have a Pomona clip and I think I may have used small pliers to bend the clip's pins to either get more of a curve or possibly straighten them out to touch the WSON's pins better.

I have some notes here: https://wiki.realmofespionage.xyz/bios:notes:rpi_flashrom

I basically dumped the BIOS from the chip, used me_cleaner to set HAP bit (didn't delete partitions), and re-flashed the me_cleaner'd BIOS back to the BIOS chip with a hardware flasher (Raspberry Pi + SOIC clip on GPIO pins; flashrom on the RPi to read/write BIOS chip). Aside from needing a patch for newer ME platforms/Coffeelake, the chip dump/flash and me_cleaner process was mostly the same as a few other Skylake boards.

___

Iirc there's something about a voltage on some BIOS chips (3.3v?) and standby voltage, or something like the motherboard supplying voltage to the BIOS chip and that not being ideal while connecting a hardware programmer to the BIOS chip at the same time. I don't recall running into that being an issue and think I flashed the chip with the laptop battery and CMOS batteries still connected (no AC adapter plugged in).

bingflash2, 2026-10-04:
can you post your ori and mod files for reference ? or a list of changes and offsets ?